kitspace/1clickBOM #3: Ninja reports error message (but everything builds and chrome seems to function)

jdar, 2016-04-19:
Hi,

Build successful and tests pass. But there was a weird error message.
Reporting this, in case you haven't seen this yourself. // or it's a platform thing.

"depfile has multiple output paths"

uname -a
15.4.0 Darwin Kernel Version 15.4.0: Fri Feb 26 22:08:05 PST 2016; root:xnu-3248.40.184~3/RELEASE_X86_64 x86_64
OSX: 10.11.4 (15E65)

Ninja 1.6.0

browserify@13.0.0

CLI History:

darius:1clickBOM jdar$ npm install
darius:1clickBOM jdar$ export PATH=$PATH:$(pwd)/node_modules/.bin
darius:1clickBOM jdar$ ./configure.coffee
generated build.ninja
darius:1clickBOM jdar$ ninja
[7/7] browserify --require='./build/.t...1.11.0.js' -o build/chrome/js/qunit.js
FAILED: echo -n 'build/chrome/js/qunit.js: ' > build/chrome/js/qunit.js.d && browserify -x  --debug --extension=".coffee" --transform coffeeify --require='./build/.temp-chrome/qunit-1.11.0.js' --list | sed 's!/Users/jdar/src/1clickBOM/!!' | tr '\n' ' ' >> build/chrome/js/qunit.js.d && browserify -x  --debug --extension=".coffee" --transform coffeeify --require='./build/.temp-chrome/qunit-1.11.0.js' -o build/chrome/js/qunit.js
[7/7] browserify build/.temp-chrome/op...s.coffee -o build/chrome/js/options.js
FAILED: echo -n 'build/chrome/js/options.js: ' > build/chrome/js/options.js.d && browserify -x  --debug --extension=".coffee" --transform coffeeify build/.temp-chrome/options.coffee --list | sed 's!/Users/jdar/src/1clickBOM/!!' | tr '\n' ' ' >> build/chrome/js/options.js.d && browserify -x  --debug --extension=".coffee" --transform coffeeify build/.temp-chrome/options.coffee -o build/chrome/js/options.js
[7/7] browserify build/.temp-chrome/popup.coffee -o build/chrome/js/popup.js
FAILED: echo -n 'build/chrome/js/popup.js: ' > build/chrome/js/popup.js.d && browserify -x  --debug --extension=".coffee" --transform coffeeify build/.temp-chrome/popup.coffee --list | sed 's!/Users/jdar/src/1clickBOM/!!' | tr '\n' ' ' >> build/chrome/js/popup.js.d && browserify -x  --debug --extension=".coffee" --transform coffeeify build/.temp-chrome/popup.coffee -o build/chrome/js/popup.js
[7/7] browserify build/.temp-firefox/p....coffee -o build/firefox/data/popup.js
FAILED: echo -n 'build/firefox/data/popup.js: ' > build/firefox/data/popup.js.d && browserify -x  --debug --extension=".coffee" --transform coffeeify build/.temp-firefox/popup.coffee --list | sed 's!/Users/jdar/src/1clickBOM/!!' | tr '\n' ' ' >> build/firefox/data/popup.js.d && browserify -x  --debug --extension=".coffee" --transform coffeeify build/.temp-firefox/popup.coffee -o build/firefox/data/popup.js
[7/7] browserify build/.temp-chrome/unit.coffee -o build/chrome/js/unit.js
FAILED: echo -n 'build/chrome/js/unit.js: ' > build/chrome/js/unit.js.d && browserify -x build/.temp-chrome/qunit-1.11.0.js --debug --extension=".coffee" --transform coffeeify build/.temp-chrome/unit.coffee --list | sed 's!/Users/jdar/src/1clickBOM/!!' | tr '\n' ' ' >> build/chrome/js/unit.js.d && browserify -x build/.temp-chrome/qunit-1.11.0.js --debug --extension=".coffee" --transform coffeeify build/.temp-chrome/unit.coffee -o build/chrome/js/unit.js
[7/7] browserify build/.temp-chrome/fu...offee -o build/chrome/js/functional.js
FAILED: echo -n 'build/chrome/js/functional.js: ' > build/chrome/js/functional.js.d && browserify -x build/.temp-chrome/qunit-1.11.0.js --debug --extension=".coffee" --transform coffeeify build/.temp-chrome/functional.coffee --list | sed 's!/Users/jdar/src/1clickBOM/!!' | tr '\n' ' ' >> build/chrome/js/functional.js.d && browserify -x build/.temp-chrome/qunit-1.11.0.js --debug --extension=".coffee" --transform coffeeify build/.temp-chrome/functional.coffee -o build/chrome/js/functional.js
[7/7] browserify build/.temp-chrome/main.coffee -o build/chrome/js/main.js
FAILED: echo -n 'build/chrome/js/main.js: ' > build/chrome/js/main.js.d && browserify -x  --debug --extension=".coffee" --transform coffeeify build/.temp-chrome/main.coffee --list | sed 's!/Users/jdar/src/1clickBOM/!!' | tr '\n' ' ' >> build/chrome/js/main.js.d && browserify -x  --debug --extension=".coffee" --transform coffeeify build/.temp-chrome/main.coffee -o build/chrome/js/main.js
depfile has multiple output paths
ninja: build stopped: subcommand failed.

kasbah, 2016-04-20:
Hey, glad you managed to compile and test and thanks for reporting. 

This is an issue with the creation of the [depfile](https://ninja-build.org/manual.html#_depfile) that I can't reproduce (on Ubuntu). It is most likely because of some subtle difference in `sed` or `tr` on OSX. 

I will switch to using [a less platform dependant script I used for Kitnic](https://github.com/monostable/kitnic/blob/master/depfileify.coffee) when I get a chance.

For the time being the dependency info won't be tracked properly so you might have some issue with the ninja re-building things that don't need to be rebuilt or not rebuilding things when they do.

kasbah, 2016-04-20:
I have pushed a fix to master. Please test.

jdar, 2016-04-20:
Yes, this eliminated the warning/error messages

Thanks for the insight.

closed 2016-04-20 by jdar