kitspace/1clickBOM #12: Capacitor Auto Match Doesn't Make Sense

borgel, 2017-10-12:
Hello!

I have a capacitor in my BOM which is described as `0.1uF 0805 X5R`. When I auto complete it, on DigiKey I seem to get this [1] which is the correct size and type, except it's a capacitor _array_ not a single cap (pretty strange part actually!). Everything else seems to be matching correctly.

Is there anything I can do to encourage it to do the right thing?

[1] http://www.digikey.com/scripts/DkSearch/dksus.dll?Detail&itemSeq=240865863&uq=636434004340247418

kasbah, 2017-10-12:
Thanks for reporting. This is unfortunately down to the [Octopart search results](https://octopart.com/search?q=0.1uF%200805%20X5R) and the fact that 1clickBOM will pick the part with the lowest MOQ out of those results. 

I am aware of these issues and built and integrated [electro-grammar](https://monostable.github.io/electro-grammar/) to address the bulk of these. Unfortunately that currently only works if there is a CPL part that matches. There _is_ an X7R CPL part that would match if you removed X5R from the description so that's what I would recommend in this case. 

Even so, the real problem with the completion as it stands is that it's like a self driving car with one button: drive! It's fine if it works but it's a pain if it goes wrong. :laughing: That is why I am building [bom-builder](https://github.com/kasbah/bom-builder), which I mentioned to you previously and sent you a screenshot of, to do this sort of completion in a more interactive way. Once that is done I will probably remove the auto-completion from 1clickBOM and just have the bom-builder on Kitnic.  I have also been talking to Octopart to help them improve their results.

borgel, 2017-10-14:
Interesting. Thanks for explaining the matching process! In this case arbitrarily specifying X7R works fine, but it's useful to know what's going on under to hood.

Awesome. Looking forward to trying it :)

Thanks for the help!

closed 2017-10-14 by borgel