kitspace/1clickBOM #14: Does this still work?

TheFlyingZephyr, 2018-03-18:
I've tried using in on Chrome and Firefox. When I click paste, a text box with contents 'war' appears over the toolbar icon. I assume this is a clipped warning. It would be great if I could read the warning. Short of that, is this project still supported or even working? It looks like it would be great if it did.

kasbah, 2018-03-18:
Hello! What OS are you using? Could you show a screenshot of what you are seeing? The warning should probably say something like "Your data doesn't look like tab-separated values". It is meant to be used in conjunction with a [spreadsheet with the right columns in it](https://github.com/monostable/1clickBOM#all-cad-programs). 

To reassure yourself that it works you could load some data from a Kitspace project e.g. [push-on-hold-off](https://kitspace.org/boards/github.com/kasbah/push-on-hold-off/) by pressing the blue button in the extension when you are on that page.

kasbah, 2018-03-18:
By the way, if you are interested in the "auto-complete" feature I have been working on an [alpha version of a BOM builder tool](https://bom-builder.kitspace.org/) that might work better for you.

kasbah, 2018-03-30:
@TheFlyingZephyr, is this issue resolved?