kitspace/1clickBOM #19: decimals autocomplete

thijsdeschildre, 2019-03-30:
I'm testing the software with autocomplete. I can make it work for 1K or 10K resistors, but not for values with a decimal. I tried entering 1.5K, 1,5k  and 1K5. Can't get it to work. How is this supposed to go?

kasbah, 2019-03-30:
Hi, thanks for trying it out and reporting. I think what you are doing should work but with the autocomplete feature it's a bit hard to see what's going on. The issue may be that it was never made to work well for through-hole components. 

I have been working on a new tool to help select parts: [the BOM Builder](https://bom-builder.kitspace.org/). You should be able to enter "1.5k 0603" or "1k5 0603" into the description column there and get back good results. Give it a go and hit me up if you have any problems.

thijsdeschildre, 2019-03-30:
oh, I'm only using SMD components. 
the BOM builder looks very nice, could be more convenient than copypasting. 

prying more into it: 1.5k 0603 works for BOM builder, but not for 1206, 0805 or 0402. clicking the magnifying glass still gives me options, sometimes the first result is a resistor array.
going back to the plugin for Chrome, I can't replicate the issue, and neither can a friend.

kasbah, 2019-03-30:
You can't replicate your initial issue? Yes, a resistor array is what Octopart returns sometimes unfortunately, we are working on improving these results on our side.

From the bom-builder you can download a csv and then paste that into the extension from a spreadsheet editor.