kitspace/1clickBOM #26: Mouser and LCSC bugs for ULX3S

emard, 2020-06-03:
HI

For our project ULX3S 
https://kitspace.org/boards/github.com/emard/ulx3s/

mouser doesn't exactly recognize part
Mouser: 700-MAX11125ATI+
and reports this (mouser actual PN is really ending with "+" if that is making problems here?)
More Than 1 Match Found for 700MAX11125ATI Select from a list of matching products

LCSC is very unreliable, I have to retry cca 10 times to get 48 parts added and 1 part
not added, but it's difficult to find which one and what is the problem about it.

kasbah, 2020-06-03:
Moved this to 1clickBOM as these are browser extension bugs.

kasbah, 2020-06-05:
@emard I spent some time investigating just now. The issue is that you two lines use the same LCSC part: C364562 and two more use the part: C375453. The LCSC website merges these parts for you, that is why you have two lines missing from your cart without any warnings.

We should try and handle this case better and at least issue some warnings from the extension. In the meantime you can avoid it by making sure you do not have duplicate LCSC parts. 

Regarding the other issue:
> (clicking LCSC, scrolling at items for few seconds,
>  emptying cart if there are any items in the cart and
> repeating click LCSC again...)
> 
> 48 of 48 lines failed
> ...

This one is hard for me to reproduce. Do you wait for the "empty cart" to finish before clicking "add to cart"? 

And this one: 
> Mouser: 700-MAX11125ATI+

Will check that now

emard, 2020-06-05:
Thanks for locating the problems!
Because board is produced in few different models with different
parts (actually this are color LEDs)

I had to use a "trick" I create 2 different schematic diode
symbols with the same part number. One symbol can change part #
for different board production, other must stay.

Mouser is "smart" here and merges the parts together, I'd like if LCSC
could do the same, or I must find a new trick how to have same part
# on 2 different symbols.

LCSC emptying cart occurs immediately but if it's only for waiting
I can try that approach

Thanx for looking what's up with
700-MAX11125ATI+ that's the problem we see for a long time

emard, 2020-06-05:
With LCSC waiting after emtpying cart doesn't help,
I emptied, closed browser and waited about 10 minutes.

After that I clicked LCSC and only 10 parts appeared (instead of
something like 46)

kasbah, 2020-06-05:
Thanks for checking. I only thought you were maybe clicking it before the empty was finished. 10 minutes is a lot, haha. 

Do you use Chrome or Firefox?

emard, 2020-06-05:
my browser is firefox-esr 68.8.0esr (64-bit) linux debian (testing)

kasbah, 2020-06-05:
I actually use Firefox 76 myself but normally develop the extension in Chrome or Chromium. I just tried in Firefox and was just able to see something similar. This may also be related to https://github.com/kitspace/kitspace/issues/243.  Will investigate further. 

The issue with Mouser an the "+" was fixed in v1.6.2, submitted for review with Chrome and Firefox extensions teams. You should get the update soon.

emard, 2020-06-05:
thanx for Mouse I'll update when I get notice :)

kasbah, 2020-06-06:
@emard I have submitted v1.7.1 for review which should fix all of your complaints and fix some other things. I added some warnings about merging lines with the same parts for e.g. LCSC. The LCSC + Firefox things turns out it was just sending too many http requests too fast and was being rate limited. Full changelog:

- Fix issue with geolocation
- Fix issues with LCSC add to cart on Firefox
- Add warnings before merging lines with same parts
- Fix issue with adding to Digikey with more than 30 line items

Going to close this, if you notice anything else going wrong, please do let me know, very much appreciated.

closed 2020-06-06 by kasbah

emard, 2020-06-06:
Thanks a lot!
I have tested LCSC and Mouser, everything works with latest v1.7.1 !
all issues are fixed :)