kitspace/1clickBOM #29: error with ulx3s project

emard, 2020-10-14:
HI

trying to generate 1-click BOM I got this error.
I pulled latest from git

Run command:
python "/home/guest/src/downloaded/1clickBOM/output_scripts/kicad/1-click-bom_fields.py" "/mt/scratch/tmp/fpga/circuits/ulx3s/ulx3s.xml" "/mt/scratch/tmp/fpga/circuits/ulx3s/ulx3s"

Command error. Return code -1
Error messages:
execvp(python, /home/guest/src/downloaded/1clickBOM/output_scripts/kicad/1-click-bom_fields.py, /mt/scratch/tmp/fpga/circuits/ulx3s/ulx3s.xml, /mt/scratch/tmp/fpga/circuits/ulx3s/ulx3s) failed with error 2!

kasbah, 2020-10-14:
Hey, I can't reproduce this locally. I get

```
Run command:
python "/home/kaspar/projects/kitspace/1clickBOM/output_scripts/kicad/1-click-bom_fields.py" "/home/kaspar/projects/kitspace/boards/ulx3s/ulx3s.xml" "/home/kaspar/projects/kitspace/boards/ulx3s/ulx3s"

Success
```

I tried it with Python2 and Python3. It generates this: https://cloud.monostable.co.uk/ulx3s-1-click-bom.tsv

emard, 2020-10-14:
Hmmm thanx for testing, I must check again what do I have locally

emard, 2020-10-14:
HI

I think I no longer have "python" , on system. Running command
manually with python3 generates BOM. Main thing works, this
kicad menu entry I can fix easily

kasbah, 2020-10-14:
I see, thanks. Let me know if you run into any other issues.

closed 2020-10-14 by kasbah

emard, 2020-10-14:
The fix is just to rename "python" -> "python3" in BOM generation menu entry
commandline